notro/rpi-firmware #15: Waveshare 3.2inch rpi lcd for the Raspberry

boaz-jan, 2015-01-18:
Hi Notro,
This is not really an issue but rather a help request so thanks in advanced.

I got the Waveshare 3.2inch rpi lcd for the Raspberry to play with and I started fiddling around with all it's GPIOs to see how to turn off the back-light as no documentation was provided by the manufacturer.
Once I switched GPIO 27 to 0 (from what I saw in another issue here, it's the RESET GPIO) the screen went white and stop responding.
What did the reset do? how can I recover from that (if I can)? 

Thanks,
Boaz

P.S, thanks for this awesome firmware!

notro, 2015-01-18:
Pulling reset low resets the LCD controller and it looses the setup. Only way to recover is reloading the driver or device.
I'm not sure if there is any backlight control: https://github.com/notro/rpi-firmware/issues/6#issuecomment-65303197

boaz-jan, 2015-01-18:
Sorry for my noobiness, I tried rebooting the pi, shouldn't it reload the driver?
Can you please elaborate a bit on how to do that? or direct me to a place that does?

Thanks very much for answering!!!

notro, 2015-01-18:
Rebooting reloads the driver. The driver briefly pulls 'reset' low during probing.
You  know the driver is loaded if you have ```/dev/fb1```

boaz-jan, 2015-01-18:
I don't have /dev/fb0
I checked dmesg and no spi were created
I followed imaginal's instructions to letter from here (https://github.com/notro/rpi-firmware/issues/6) and it worked perfectly well until I changed 27 to low.

Any help will be greatly appropriated

Thanks!

notro, 2015-01-18:
Ok, that kernel has the drivers built into the kernel proper (kernel.img)
This should show you if the drivers are present:
```
ls -l /sys/bus/spi/drivers
```

If you don't have any fbtft_device entries in the boot log, the driver won't work because the device is not present.
```
dmesg | grep fbtft_device
```

Your fbtft_device entries from /boot/cmdline.txt determines if fbtft_device actually does something. They should show up here:
```
cat /proc/cmdline
```

boaz-jan, 2015-01-18:
Well, the first command gave the answer.
It seems that to install the gpio command from here http://elinux.org/RPi_Low-level_peripherals#Shell_script_-_take_2 required me to update some packages on the OS.
Apparently the update also removed some of said drivers, but running the rpi-update again said I have the latest version.
I remove /boot/.firmware_revision and reinstalled the executed again the rpi-update command and now everything works!!

Thanks Notro for your help and hard work!!

closed 2015-01-18 by boaz-jan