notro/rpi-firmware #16: Raspberry pi 2 support

locusf, 2015-02-05:
Hello,

Does this firmware first of all support:

* PiScreen
* Raspberry pi 2

I could wait if you want to make the support yourself, just refer this issue.

notro, 2015-02-05:
I added and tested piscreen support some time ago: https://github.com/notro/fbtft/blob/master/fbtft_device.c#L679

I just received a Pi 2, and hope to look into it this evening.

cm-t, 2015-02-05:
waiting for updates notro !
*I will receive my Pi2 in something like 15 days, with the hope it will works…*

notro, 2015-02-05:
With a fix to the SPI DMA driver, I have FBTFT working on the Pi 2.
Tomorrow I'll change my build scripts to handle this 2 kernel situation (Pi1: kernel.img and Pi2: kernel7.img).

notro, 2015-02-07:
Pi2 support is added in the latest firmware. Please try it.
I have added some text here: https://github.com/notro/fbtft/wiki#install

locusf, 2015-02-13:
I still can't make the piscreen work yet, it seems that the device isn't even powering up at all and no devices for the touchscreen can be found.

notro, 2015-02-14:
I suggest you contact ozzmaker about this.
The overlay was testet with a PiScreen I have from one of the first batches.

locusf, 2015-02-14:
Yes I got it working, I just needed to add dtoverlay=piscreen to config.txt instead of cmdline.txt. lol

closed 2015-02-15 by locusf