notro/rpi-firmware #19: pi 2 kernel panics after install this driver for a JBTEK 3.5" LCD

acidjazz, 2015-02-26:
After I run

```bash
pi@raspberrypi ~ $sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update
```

And reboot, I get:

[image]

acidjazz, 2015-02-26:
After a second reboot:

[image]

acidjazz, 2015-02-26:
Here is the result from repeating the top steps but not running rpi-update beforehand:

[image]

prodigy7, 2015-02-26:
Same here too! Kernel panic!

acidjazz, 2015-02-26:
I was able to get the kernel to panic to not shut down the PI 2
```dmesg

[   15.594330] cfg80211: Calling CRDA to update world regulatory domain
[   16.170506] cfg80211: Calling CRDA to update world regulatory domain
[   17.146919] grow_buffers: requested out-of-range block 56517474700441 for device mmcblk0p2
[   17.146980] Unable to handle kernel paging request at virtual address 2b5520a0
[   17.160334] pgd = b9384000
[   17.168621] [2b5520a0] *pgd=00000000
[   17.175866] Internal error: Oops: 5 [#1] PREEMPT SMP ARM
[   17.184932] Modules linked in: fbtft_device flexfb fbtft syscopyarea sysfillrect sysimgblt fb_sys_fops snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd arc4 rt2800usb rt2800lib rt2x00usb rt2x00lib mac80211 cfg80211 crc_ccitt rfkill spi_bcm2708 evdev uio_pdrv_genirq uio
[   17.217790] CPU: 3 PID: 2100 Comm: swapon Not tainted 3.18.5-v7+ #1
[   17.227593] task: b916cac0 ti: b8340000 task.ti: b8340000
[   17.236532] PC is at __brelse+0x14/0x60
[   17.243954] LR is at ext4_ext_drop_refs+0x3c/0x50
[   17.252273] pc : [<80176430>]    lr : [<801ecab8>]    psr: 20000013
[   17.252273] sp : b8341b78  ip : b8341b88  fp : b8341b84
[   17.270830] r10: b921bec0  r9 : 00000020  r8 : 00000000
[   17.279564] r7 : 00000000  r6 : 00000004  r5 : 00000003  r4 : b921bf00
[   17.289615] r3 : 00000000  r2 : 00000001  r1 : ba07e3c0  r0 : 2b552068
[   17.299698] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   17.310379] Control: 10c5387d  Table: 3938406a  DAC: 00000015
[   17.319702] Process swapon (pid: 2100, stack limit = 0xb8340238)
[   17.329327] Stack: (0xb8341b78 to 0xb8342000)
[   17.337288] 1b60:                                                       b8341ba4 b8341b88
[   17.349115] 1b80: 801ecab8 80176428 fffffff4 00003367 00001d4b 00000003 b8341c04 b8341ba8
[   17.361014] 1ba0: 801ecf98 801eca88 0000cc99 00003367 00000003 00000000 b8341be4 b8341bc8
[   17.372794] 1bc0: 800555f8 b9731f80 00000000 00000000 00000000 b9732018 b8341c44 b9732018
[   17.384523] 1be0: b8341d60 00000000 00000000 00001d4b 00000001 00000000 b8341cdc b8341c08
[   17.396254] 1c00: 801f0f14 801eccf8 b8341c24 b8341c18 b8341c2c b8341c20 8020401c 80204620
[   17.408036] 1c20: 00001d49 00000000 80204f60 80203f98 b95da920 b95da904 b95da900 00000000
[   17.419859] 1c40: 00000000 b8341c98 b9a29c00 b8341c58 80204738 802ed4ac 000050b0 00000000
[   17.431704] 1c60: b8341c7c 00000000 00000000 b973219c b8341c98 b9732018 b9731f80 00001d4a
[   17.443567] 1c80: b8341d60 8020558c b8341cdc b8341c98 8020558c 8054fc58 b8341cdc b8341ca8
[   17.455432] 1ca0: 801f0cc8 8020580c b8341cdc b8341cb8 8020580c 00000000 00000000 b9732018
[   17.469129] 1cc0: 00000000 00000000 b8341d60 00000000 b8341d4c b8341ce0 801c6828 801f0eb8
[   17.482840] 1ce0: 000050b0 00000000 00000004 00000003 b8341d3c 80550bac b8341da4 00000001
[   17.496524] 1d00: 80550bb0 800d2e14 00000000 00000000 00000000 00000020 00000000 00000000
[   17.510176] 1d20: b9732018 00000000 00000000 b9732018 b8341dc8 00000000 00000000 b920c0c0
[   17.523843] 1d40: b8341da4 b8341d50 801c6bdc 801c6664 b8341d84 b8341d60 80075658 00000000
[   17.537547] 1d60: 000050b0 00000000 00001d4b 00000001 00000000 80803ac0 b8341d9c 00000000
[   17.551295] 1d80: 00000000 b9732018 b97320ec 00001d4b b9732018 b920c0c0 b8341dbc b8341da8
[   17.565091] 1da0: 801c6d24 801c6b54 b8341dc8 00000000 b8341e1c b8341dc0 801744b0 801c6d04
[   17.578918] 1dc0: b8341dc8 00000000 00000000 00000003 b8341e3c 80550bac 00000000 00000000
[   17.592792] 1de0: 00001000 8015cd00 b9422400 00000003 b8341e3c 80550bac b8341e5c b8341e08
[   17.606686] 1e00: b8341e1c b8341e10 00001d4b 00000000 b8341e4c b8341e20 801c4a3c 80174468
[   17.620631] 1e20: 801c6cf8 00000000 00001a0d 00001d4b 00000000 000050b0 00000000 00001d4b
[   17.634622] 1e40: b8341e5c b8341e50 8015cd20 801c49a8 b8341f04 b8341e60 8012dd94 8015cd00
[   17.648679] 1e60: 000050b0 00000000 00003367 00000000 000050b0 00000000 000050b0 00000000
[   17.662817] 1e80: b8341f70 00000001 00000008 801127ac 00000001 00000001 801127ac 8054dee8
[   17.676975] 1ea0: 0000000c 00000001 b8341ebc 00009a04 00000000 00000000 00000001 00000000
[   17.691164] 1ec0: 00006400 00000000 00000001 b920c0c0 00001d4c 00000000 bd890000 b920c0c0
[   17.705321] 1ee0: 00006400 b9732018 bd890000 b920c6c0 808f2d64 00006400 b8341fa4 b8341f08
[   17.719454] 1f00: 80130c78 8012dc60 80163284 80163090 b8341f54 b8341f20 801467e4 8016325c
[   17.733578] 1f20: 00000020 00000000 b920c000 b916cee4 00000000 8086c05c b916cac0 000058ac
[   17.747719] 1f40: 00000064 00000000 b8341f64 000063ff 000063ff b920c128 00006400 b8340008
[   17.761860] 1f60: bd888000 00000000 b9033000 ba764680 b8340000 00001000 0082e048 00001000
[   17.775995] 1f80: 0082e048 00000003 00000057 8000f964 b8340000 00000000 00000000 b8341fa8
[   17.790152] 1fa0: 8000f6e0 8012ffb4 00001000 0082e048 0082e038 00000000 00000000 00000001
[   17.804332] 1fc0: 00001000 0082e048 00000003 00000057 06400000 00000000 00010000 0082e038
[   17.818436] 1fe0: 00001000 7ef53bb0 0000acb4 76e8003c 60000010 0082e038 00000000 00000000
[   17.832507] [<80176430>] (__brelse) from [<801ecab8>] (ext4_ext_drop_refs+0x3c/0x50)
[   17.846150] [<801ecab8>] (ext4_ext_drop_refs) from [<801ecf98>] (ext4_find_extent+0x2ac/0x330)
[   17.860724] [<801ecf98>] (ext4_find_extent) from [<801f0f14>] (ext4_ext_map_blocks+0x68/0x15f0)
[   17.875415] [<801f0f14>] (ext4_ext_map_blocks) from [<801c6828>] (ext4_map_blocks+0x1d0/0x4f0)
[... cut ...]

```
flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3
 fbtft_device debug=3 rotate=90 name=flexfb speed=16000000 gpios=reset:25,dc:24

```

acidjazz, 2015-02-26:
Here is my syslog as well from the above dump

```syslog
eb 26 21:08:40 raspberrypi kernel: [    4.924340] random: nonblocking pool is initialized
Feb 26 21:08:40 raspberrypi kernel: [    5.206612] cfg80211: Calling CRDA to update world regulatory domain
Feb 26 21:08:40 raspberrypi kernel: [    5.389870] usb 1-1.2: reset high-speed USB device number 4 using dwc_otg
Feb 26 21:08:40 raspberrypi kernel: [    5.528989] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 5592, rev 0222 detected
Feb 26 21:08:40 raspberrypi kernel: [    5.564354] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 000f detected
Feb 26 21:08:40 raspberrypi kernel: [    5.603782] usbcore: registered new interface driver rt2800usb
Feb 26 21:08:40 raspberrypi kernel: [    7.127310] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Feb 26 21:08:40 raspberrypi kernel: [    7.458169] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Feb 26 21:08:40 raspberrypi kernel: [    8.127410] fbtft_device:  SPI devices registered:
Feb 26 21:08:40 raspberrypi kernel: [    8.138419] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
Feb 26 21:08:40 raspberrypi kernel: [    8.150868] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
Feb 26 21:08:40 raspberrypi kernel: [    8.163277] fbtft_device:  'fb' Platform devices registered:
Feb 26 21:08:40 raspberrypi kernel: [    8.174950] fbtft_device:      bcm2708_fb id=-1 pdata? no
Feb 26 21:08:40 raspberrypi kernel: [    8.186415] fbtft_device: Deleting spi0.0
Feb 26 21:08:40 raspberrypi kernel: [    8.197544] flexfb spi0.0: fbtft_request_gpios: 'reset' = GPIO25
Feb 26 21:08:40 raspberrypi kernel: [    8.209903] flexfb spi0.0: fbtft_request_gpios: 'dc' = GPIO24
Feb 26 21:08:40 raspberrypi kernel: [    8.221572] flexfb spi0.0: flexfb_verify_gpios_dc()
Feb 26 21:08:40 raspberrypi kernel: [    8.232200] flexfb spi0.0: fbtft_init_display()
Feb 26 21:08:40 raspberrypi kernel: [    8.242365] flexfb spi0.0: fbtft_reset()
Feb 26 21:08:40 raspberrypi kernel: [    8.372084] flexfb spi0.0: init: write(0xB0) 0x00
Feb 26 21:08:40 raspberrypi kernel: [    8.382788] flexfb spi0.0: init: write(0x11)
Feb 26 21:08:40 raspberrypi kernel: [    8.393702] flexfb spi0.0: init: mdelay(250)
Feb 26 21:08:40 raspberrypi kernel: [    8.654205] flexfb spi0.0: init: write(0x3A) 0x55
Feb 26 21:08:40 raspberrypi kernel: [    8.664644] flexfb spi0.0: init: write(0xC2) 0x44
Feb 26 21:08:40 raspberrypi kernel: [    8.674922] flexfb spi0.0: init: write(0xC5) 0x00 0x00 0x00 0x00
Feb 26 21:08:40 raspberrypi kernel: [    8.686510] flexfb spi0.0: init: write(0xE0) 0x0F 0x1F 0x1C 0x0C 0x0F 0x08 0x48 0x98 0x37 0x0A 0x13 0x04 0x11 0x0D 0x00
Feb 26 21:08:40 raspberrypi kernel: [    8.703039] flexfb spi0.0: init: write(0xE1) 0x0F 0x32 0x2E 0x0B 0x0D 0x05 0x47 0x75 0x37 0x06 0x10 0x03 0x24 0x20 0x00
Feb 26 21:08:40 raspberrypi kernel: [    8.719474] flexfb spi0.0: init: write(0xE2) 0x0F 0x32 0x2E 0x0B 0x0D 0x05 0x47 0x75 0x37 0x06 0x10 0x03 0x24 0x20 0x00
Feb 26 21:08:40 raspberrypi kernel: [    8.735798] flexfb spi0.0: init: write(0x36) 0x28
Feb 26 21:08:40 raspberrypi kernel: [    8.745915] flexfb spi0.0: init: write(0x11)
Feb 26 21:08:40 raspberrypi kernel: [    8.755451] flexfb spi0.0: init: write(0x29)
Feb 26 21:08:40 raspberrypi kernel: [    8.933262] flexfb spi0.0: Display update: 1782 kB/s (168.294 ms), fps=0 (0.000 ms)
Feb 26 21:08:40 raspberrypi kernel: [    8.946159] flexfb spi0.0: fbtft_register_backlight()
Feb 26 21:08:40 raspberrypi kernel: [    8.956215] flexfb spi0.0: fbtft_register_backlight(): led pin not set, exiting.
Feb 26 21:08:40 raspberrypi kernel: [    8.969122] graphics fb1: flexfb frame buffer, 480x320, 300 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 16 MHz
Feb 26 21:08:40 raspberrypi kernel: [    8.985699] fbtft_device:  GPIOS used by 'flexfb':
Feb 26 21:08:40 raspberrypi kernel: [    8.995317] fbtft_device:    'reset' = GPIO25
Feb 26 21:08:40 raspberrypi kernel: [    9.004296] fbtft_device:    'dc' = GPIO24
Feb 26 21:08:40 raspberrypi kernel: [    9.012878] fbtft_device:  SPI devices registered:
Feb 26 21:08:40 raspberrypi kernel: [    9.022025] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
Feb 26 21:08:40 raspberrypi kernel: [    9.032644] fbtft_device:      flexfb spi0.0 16000kHz 8 bits mode=0x00
Feb 26 21:08:40 raspberrypi kernel: [   12.942598] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
Feb 26 21:08:40 raspberrypi kernel: [   12.946183] ieee80211 phy0: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.29
Feb 26 21:08:40 raspberrypi kernel: [   13.495181] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Feb 26 21:08:40 raspberrypi kernel: [   15.594330] cfg80211: Calling CRDA to update world regulatory domain
Feb 26 21:08:40 raspberrypi kernel: [   16.170506] cfg80211: Calling CRDA to update world regulatory domain
Feb 26 21:08:42 raspberrypi kernel: [   17.184932] Modules linked in: fbtft_device flexfb fbtft syscopyarea sysfillrect sysimgblt fb_sys_fops snd_bcm2835 snd_pcm snd_seq snd_seq_device snd_timer snd arc4 rt2800usb rt2800lib rt2x00usb rt2x00lib mac80211 cfg80211 crc_ccitt rfkill spi_bcm2708 evdev uio_pdrv_genirq uio
Feb 26 21:08:42 raspberrypi kernel: [   17.217790] CPU: 3 PID: 2100 Comm: swapon Not tainted 3.18.5-v7+ #1
Feb 26 21:08:42 raspberrypi kernel: [   17.227593] task: b916cac0 ti: b8340000 task.ti: b8340000
Feb 26 21:08:42 raspberrypi kernel: [   17.236532] PC is at __brelse+0x14/0x60
[... cut ...]
Feb 26 21:08:57 raspberrypi kernel: [   32.956764] wlan0: RX AssocResp from 74:d0:2b:5e:7c:c0 (capab=0x411 status=0 aid=2)
Feb 26 21:08:57 raspberrypi kernel: [   32.981339] wlan0: associated
```

Maybe the latest firmware from [rpi-firmware](https://github.com/Hexxeh/rpi-firmware/commit/0c9001b568d1ac9593925fb76795bcc25cc59558) which specifys kernel-level fixes might help? I don't really know. Maybe an overlay for the JBTEK? it seems that 10% of the time now I can boot and send an image to the screen, otherwise I kernel panic

notro, 2015-02-26:
It seems you have a wifi dongle or something. Does it panic without that attached?
Have you tried loading the fbtft modules manually after boot?

prodigy7, 2015-02-26:
Made a fresh clean installation for a new rpi2 without any device attached. Only display, network and keyboard are connected. I think, attached devices are not the reason.

notro, 2015-02-26:
There will be [fbtft support](https://github.com/raspberrypi/linux/pull/836) in the next official Pi kernel build.
Keep an eye here to spot when it's released: https://github.com/Hexxeh/rpi-firmware/commits/master
Update with plain ```rpi-update``` and see if that works.
It doesn't have SPI DMA support, so I'll make a new fbtft kernel release when I'm done with all the DT overlays.

notro, 2015-02-26:
This is not clear to me: Does the kernel panic also when the fbtft drivers is not loaded?

acidjazz, 2015-02-26:
I tried unattaching various things, the only times it panics is when the LCD is plugged in no matter what else is attached

acidjazz, 2015-02-26:
When the LCD is attached, about 5% of the time it boots it works and I can send things to the LCD, the other times I get kernel panics.  I do though see the screen turn black and see the fbtft driver load during boot, which means its activating.  I just unforutnately don't get to a command prompt w/out it crashing

acidjazz, 2015-02-26:
Thanks @notro I'll definitely keep my eye out for that.  Glad to see your work in the official kernel.

notro, 2015-02-26:
> I tried unattaching various things, the only times it panics is when the LCD is plugged in no matter what else is attached

And the driver is still loaded/probed when the LCD is unplugged?
The reason I ask is that the driver doesn't know if there is hardware attached or not. The SPI bus emits it's signals as long as the driver is loaded and something is driving /dev/fb1.
Do you have a Pi1 to test with?

acidjazz, 2015-02-27:
> And the driver is still loaded/probed when the LCD is unplugged?
I don't touch the drivers at all, i only plug in the LCD for a 95% crash, and unplug it for a 100% boot. So if the driver still loads than it's not crashing when the driver is loaded and the LCD is not attached.

  I do have about 5 pi 1's I use the LCD with and they all work fine, but I haven't updated them with the newer kernel.  Their kernel is about a month old

prodigy7, 2015-02-27:
Is there any way how I can get a working kernel without compiling it self?

acidjazz, 2015-02-28:
I just ran rpi-update to grab the [latest](https://github.com/Hexxeh/rpi-firmware/commit/8f6196d9e3c96915d89318821c5e591f094561e5) 2.8.18 kernel with built in FBTFT drivers and my LCD is working perfect.  I've rebooted 10+ times and still can't get a crash.  I think this might have fixed it

notro, 2015-02-28:
Good, let's hope it stays that way :-)

prodigy7, 2015-02-28:
Upgraded to "vanilla" raspberry kernel but have some problems. Using a board called CCD from busware and the transmitter doesn't work (but is recognized).

With working (notro) kernel following modules are loaded on my pi:
```
rtc_ds1307             10371  0
rfcomm                 59861  10
bnep                   14705  2
evdev                  10713  1  <<
joydev                  9257  0  <<
ads7846                 8720  0  <<
ads7846_device          6125  0  <<
lirc_rpi                7502  0  <<
lirc_dev               13387  1 lirc_rpi
rc_core                22301  1 lirc_dev
snd_bcm2835            19496  0
fb_ili9341              4130  1
fbtft_device           34686  0
fbtft                  30834  2 fb_ili9341,fbtft_device
syscopyarea             2999  1 fbtft
sysfillrect             3294  1 fbtft
sysimgblt               2124  1 fbtft
fb_sys_fops             1435  1 fbtft
snd_soc_bcm2708_i2s     6186  0  <<
regmap_mmio             2818  1 snd_soc_bcm2708_i2s <<
snd_soc_core          146249  1 snd_soc_bcm2708_i2s <<
snd_compress           12594  1 snd_soc_core <<
btusb                  20247  0
regmap_i2c              1661  1 snd_soc_core
snd_pcm_dmaengine       5505  1 snd_soc_core <<
regmap_spi              1913  1 snd_soc_core <<
snd_pcm                83853  3 snd_bcm2835,snd_soc_core,snd_pcm_dmaengine
bluetooth             349704  24 bnep,btusb,rfcomm
snd_page_alloc          5140  1 snd_pcm <<
snd_seq                55476  0
rfkill                 19784  2 bluetooth
snd_seq_device          6465  1 snd_seq
snd_timer              20981  2 snd_pcm,snd_seq
leds_gpio               2079  0  <<
led_class               4373  1 leds_gpio <<
snd                    62252  7 snd_bcm2835,snd_soc_core,snd_timer,snd_pcm,snd_seq,snd_seq_device,snd_compress
spi_bcm2708             8111  0
i2c_bcm2708             4943  0
```
(marked missing modules with << )

Raspberry vanilla kernel:
```
rtc_ds1307             10351  0
rfcomm                 37920  10
bnep                   11433  2
lirc_dev               11060  0
rc_core                23526  1 lirc_dev
snd_bcm2835            21157  0
snd_pcm                90778  1 snd_bcm2835
snd_seq                61097  0
snd_seq_device          7209  1 snd_seq
snd_timer              23007  2 snd_pcm,snd_seq
snd                    66285  5 snd_bcm2835,snd_timer,snd_pcm,snd_seq,snd_seq_device
fb_ili9341              4404  1
fbtft_device           34613  0
fbtft                  34434  2 fb_ili9341,fbtft_device
syscopyarea             3153  1 fbtft
sysfillrect             3584  1 fbtft
sysimgblt               2278  1 fbtft
fb_sys_fops             1565  1 fbtft
ecb                     2279  1
btusb                  21828  0
i2c_bcm2708             6200  0
spi_bcm2708             6018  0
bluetooth             320752  24 bnep,btusb,rfcomm
rfkill                 22347  2 bluetooth
uio_pdrv_genirq         3666  0
uio                     9897  1 uio_pdrv_genirq
```
Any difference in compile options maybe?

notro, 2015-02-28:
Some modules might be built-in (kernel.img) in one and loadable (lsmod) in the other.
If I'm not mistaken, this would give you the builtin and loadable modules present:
```
cat /lib/modules/$(uname -r)/modules.builtin
cat /lib/modules/$(uname -r)/modules.order
```

HugoAdrian, 2015-05-29:
The first time you install Xubuntu was a success, but the second I get this error in the Raspberry Pi b 2 in the installation of Xubuntu of https://wiki.ubuntu.com/ARM/RaspberryPi following the video https://www.youtube.com/watch?v=UGSQ7nzVCs4 in the last step, but not my desktop image below appears, help    
[image]

notro, 2015-05-29:
You have to post this in the forum thread mentioned in the wiki you refer to.

closed 2015-09-06 by notro