notro/rpi-firmware #20: AdaFruit RA8875 board and TFT only showing white screen

elliottcarlson, 2015-03-19:
Hi,

 I have a reclaimed 4.3" TFT, 40 pin model FPC4304009 that I pulled from a MP4 player I got for free a while back. I have not been able to find a datasheet for it, but according to an aliexpress listing for it (http://www.aliexpress.com/item/4-3-4-3inch-TFT-LCD-LCM-Display-PANEL-SCREEN-RGB-480-272-FPC4304005-FPC4304006-FPC4304009/32272567308.html), it runs on the following drivers: HX8257  ILI6480 OTM5180

 I have attached it to a AdaFruit RA8875 driver board (http://www.adafruit.com/products/1590) and successfully used their library (https://github.com/adafruit/Adafruit_RA8875) from a duemilanove using the init of RA8875_800x480.

 I am now attempting to attach it to an original model RPi with the following wiring:

RA8875 | RPi | RPi Pin #
------------ | ------------- | -------------
VIN | 5V | 2
GND | GND | 6
SCK | SCLK | 23
MISO | MISO | 21
MOSI | MOSI | 19
CS | SP10 CE0 | 24
RST | GPIO 25 | 22

 Powering the RPi on causes a momentary flicker of the screen, much like when powered by an arduino. Once the RPi has booted, and I run `sudo modprobe fbtft_device name=er_tftm070_5`, the screen turns on with a solid white background. Any other interaction with the fb device fails. 

 Here is the output with 'debug=1':

> [  121.330825] fbtft_device:  SPI devices registered:
> [  121.330879] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
> [  121.330897] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
> [  121.330909] fbtft_device:  'fb' Platform devices registered:
> [  121.330955] fbtft_device:      bcm2708_fb id=-1 pdata? no
> [  121.331024] fbtft_device: Deleting spi0.0
> [  121.339279] fbtft_device:  GPIOS used by 'er_tftm070_5':
> [  121.339319] fbtft_device:    'reset' = GPIO25
> [  121.339332] fbtft_device:    'dc' = GPIO24
> [  121.339343] fbtft_device:  SPI devices registered:
> [  121.339360] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
> [  121.339380] fbtft_device:      fb_ra8875 spi0.0 5000kHz 8 bits mode=0x03
> [  121.394185] fb_ra8875 spi0.0: fbtft_request_gpios: 'reset' = GPIO25
> [  121.394246] fb_ra8875 spi0.0: fbtft_request_gpios: 'dc' = GPIO24
> [  122.870529] graphics fb1: fb_ra8875 frame buffer, 800x480, 750 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 5 MHz

Here is the output with 'debug=7':

> [   52.892755] fbtft_device:  SPI devices registered:
> [   52.892805] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
> [   52.892825] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
> [   52.892837] fbtft_device:  'fb' Platform devices registered:
> [   52.892883] fbtft_device:      bcm2708_fb id=-1 pdata? no
> [   52.892954] fbtft_device: Deleting spi0.0
> [   52.900654] fbtft_device:  GPIOS used by 'er_tftm070_5':
> [   52.900694] fbtft_device:    'reset' = GPIO25
> [   52.900707] fbtft_device:    'dc' = GPIO24
> [   52.900719] fbtft_device:  SPI devices registered:
> [   52.900735] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
> [   52.900750] fbtft_device:      fb_ra8875 spi0.0 5000kHz 8 bits mode=0x03
> [   52.956254] fb_ra8875 spi0.0: fbtft_request_gpios_match('reset')
> [   52.956342] fb_ra8875 spi0.0: fbtft_request_gpios: 'reset' = GPIO25
> [   52.956366] fb_ra8875 spi0.0: fbtft_request_gpios_match('dc')
> [   52.956405] fb_ra8875 spi0.0: fbtft_request_gpios: 'dc' = GPIO24
> [   52.956427] fb_ra8875 spi0.0: fbtft_verify_gpios()
> [   52.956450] fb_ra8875 spi0.0: init_display()
> [   52.956468] fb_ra8875 spi0.0: display size 800x480
> [   52.956484] fb_ra8875 spi0.0: fbtft_reset()
> [   53.103566] fb_ra8875 spi0.0: write_reg8_bus8: 88 0b
> [   53.103618] fb_ra8875 spi0.0: write_spi(len=2): 80 88
> [   53.103907] fb_ra8875 spi0.0: write_spi(len=2): 00 0b
> [   53.104147] fb_ra8875 spi0.0: write_reg8_bus8: 89 02
> [   53.104180] fb_ra8875 spi0.0: write_spi(len=2): 80 89
> [   53.104406] fb_ra8875 spi0.0: write_spi(len=2): 00 02
> [   53.118032] fb_ra8875 spi0.0: write_reg8_bus8: 10 0c
> [   53.118090] fb_ra8875 spi0.0: write_spi(len=2): 80 10
> [   53.118378] fb_ra8875 spi0.0: write_spi(len=2): 00 0c
> [   53.118612] fb_ra8875 spi0.0: write_reg8_bus8: 04 81
[... 264 lines cut ...]
> [   54.401895] fb_ra8875 spi0.0: fbtft_write_spi(len=4093): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
> [   54.408757] fb_ra8875 spi0.0: fbtft_write_spi(len=4093): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
> [   54.415728] fb_ra8875 spi0.0: fbtft_write_spi(len=4093): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
> [   54.422645] fb_ra8875 spi0.0: fbtft_write_spi(len=4093): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
> [   54.429502] fb_ra8875 spi0.0: fbtft_write_spi(len=4093): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
> [   54.436360] fb_ra8875 spi0.0: fbtft_write_spi(len=4093): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
> [   54.443194] fb_ra8875 spi0.0: fbtft_write_spi(len=4093): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
> [   54.450030] fb_ra8875 spi0.0: fbtft_write_spi(len=4093): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
> [   54.457043] fb_ra8875 spi0.0: fbtft_write_spi(len=4093): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
> [   54.464047] fb_ra8875 spi0.0: fbtft_write_spi(len=4093): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
> [   54.471013] fb_ra8875 spi0.0: fbtft_write_spi(len=4093): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
> [   54.477866] fb_ra8875 spi0.0: fbtft_write_spi(len=2797): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
> [   54.482574] fb_ra8875 spi0.0: Display update: 566 kB/s (1322.391 ms), fps=0 (0.000 ms)
> [   54.482621] fb_ra8875 spi0.0: fbtft_register_backlight()
> [   54.482642] fb_ra8875 spi0.0: fbtft_register_backlight(): led pin not set, exiting.
> [   54.483806] graphics fb1: fb_ra8875 frame buffer, 800x480, 750 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 5 MHz

Lastly, here is an image of the setup:
[image]

Hopefully @Pfannex still checks these issues as well (thanks for all the work you guys already have done on fbtft and his work on the RA8875 driver - read through the whole history and development of it to see if I could figure anything out on my own - alas I couldn't).

Thanks,

Elliott

elliottcarlson, 2015-03-19:
Just for completeness sake, I just received my RPi2 today - i did a fresh install, added the rpi-firmware, and did a similar hookup to the RPi2 - same result. I assume it's an issue with the driver/RA8875 - but I could just be doing something completely wrong.

Thanks again,

Elliott

notro, 2015-03-24:
White display means it's not initialized.
The cause of this is usually either wrong wiring or wrong init sequence.
(debug=3 is enough to see the init sequence)

mitjad, 2015-07-21:
Hi, did you find an error, I have the same issue?
Also, are you sure about the wiring, especially :
CS	SP10 CE0 24 (is this SPI_CE0_N - http://www.element14.com/community/servlet/JiveServlet/previewBody/73950-102-4-309126/GPIO_Pi2.png)
RST	GPIO 25	22 (GPIO_GEN6)
Thanks

razofra, 2015-08-23:
Hi, I've got the same problem. My 5" LCD display is white. I don't know how to install the driver correctly. I use Adafruit RA8875 driver board, and this LCD:
http://www.buydisplay.com/default/5-tft-lcd-display-module-wvga-800x480-high-resolution-for-mp4-gps

My wiring is:
RA8875   |  RPi
Vin        -> 5.0V
GND     -> GND
SCK      -> GPIO11 (SPI_CLK)
MISO    -> GPIO9 (SPI_MISO)
MOSI    -> GPIO10 (SPI_MOSI)
CS        -> GPIO7 (SPI_CE1_N)
RST      -> GPIO25

Can any help me?

Thanks

closed 2024-09-29 by notro