notro/rpi-firmware #23: Rpi-Update git Error

Kalakalak, 2015-04-22:
When i execute the command:
sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update
it returns the error "no hash received from github:" and then the url 
Can I somehow clone the repo and set REPO_URI=localhost/home/pi/git or bypass the hash verification?
The Hash-Verification is screwing up the installation.

notro, 2015-04-22:
It is probably temporary.

Kalakalak, 2015-04-22:
I hope so

Kalakalak, 2015-04-24:
So i finally got the firmware update working:
had to copy rpi-update to /home/pi
change the REPO_URI=https://github.com/notro/rpi-firmware in the file directly
and execute sudo UPDATE_SELF=0 /home/pi/rpi-update
that worked.

Another issue:
i installed all the tools and the driver now, but my waveshare32b running on the raspberry pi 1 b (not b+) still has a white screen.

ls /dev shows no fb1 device

Help would be appreciated

thx,

Kalakalak

notro, 2015-04-25:
> ls /dev shows no fb1 device

Then the driver is not loaded.
Maybe this can be of help https://github.com/notro/fbtft/issues/181

Kalakalak, 2015-04-26:
Thanks for the link,

i installed the builtin version - it worked and the modprobe somehow didnt load the drivers.

So how do i get retropie to run on the waveshare32b without fbcp?

Kalakalak

Kalakalak, 2015-05-06:
Nothing?

notro, 2015-05-07:
The builtin image is discontinued and won't be updated. Here is info on how to use it: https://github.com/notro/rpi-firmware/wiki/branch:-builtin

waveshare must be a mess of a display (poorly supported). No display has had so much questions in the fbtft issue tracker. I don't know retropie.

You can't run opengl based emulators without fbcp: https://learn.adafruit.com/running-opengl-based-games-and-emulators-on-adafruit-pitft-displays/overview

closed 2015-09-06 by notro