notro/rpi-firmware #24: XPT2046 Touch Controller Driver

Dinsmoor, 2015-05-28:
This is another case of Chinese manufacturers packaging custom raspbian images on CDs in order to get the touchscreen to work.

People have written driver examples, but I'm still a baby, so I haven't worked out how to implement them.

However, the controller IC listed here: http://www.logictoyz.com/electronic-components/xpt2046-tssop-16-touch-screen-controller.html states that XPT2046 can replace ADS7843E TSC2046 ADS7843 ADS7846 RSM1843 chipsets, so might I be able to use the ADS7846 driver available to use the touchscreen?

Also, here: http://www.mikroe.com/forum/viewtopic.php?f=72&t=53927 and here: https://github.com/linupi/RaspPiTouchLCD/blob/master/touch.c somebody wrote something for the touch pad controller, but still I'm not sure what to do with it.

notro, 2015-05-28:
> so might I be able to use the ADS7846 driver available to use the touchscreen?

Yes, several displays uses that controller and it works with the ads7846 driver.

https://github.com/raspberrypi/linux/blob/rpi-3.18.y/arch/arm/boot/dts/ads7846-overlay.dts
https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L133

Dinsmoor, 2015-05-31:
I figured it out. Just had to read the latest documentation for loading modules via the device tree.

The manufacturers are shipping old docs that made it rather confusing. 

Thanks.

closed 2015-05-31 by Dinsmoor