notro/rpi-firmware #28: tm022hdh26 or ili9431 not work

Jakkawad, 2015-10-10:
I use Model B+ on Jessie
1. Open SPI
2. sudo nano /etc/modprobe.d/raspi-blacklist.conf -> #blacklist spi-bcm2708
3. sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update
4. /boot/config.txt ->  dtparam=spi=on 
5. sudo nano /etc/modules -> fbtft_device name=tm022hdh26 gpios=dc:24,reset:25,led:18 rotate=90
6. /usr/share/X11/xorg.conf.d/99-fbturbo.conf ->  fb1
7. /boot/cmdline.txt ->  fbcon=map:10
It not work for me ,Thank you

I'm trying 4 day

notro, 2015-10-14:
The rpi-firmware kernel is deprecated now, the official Pi kernel has all the necessary pieces.

1. sudo rpi-update
2. /boot/config.txt -> dtparam=spi=on 
3. sudo nano /etc/modules -> fbtft_device name=tm022hdh26 gpios=dc:24,reset:25,led:18 rotate=90
4. /usr/share/X11/xorg.conf.d/99-fbturbo.conf -> fb1
5. /boot/cmdline.txt -> fbcon=map:10

Now you should have /dev/fb1
If it doesn't show up, use **dmesg** to look for errors in the kernel log.
Also try loading fbtft_device manually from the command line.

Jakkawad, 2015-10-14:
dimes = Failed to start Load Kernel Modules.

but i'm try sudo mod probe fbtft_device name=tm022hdh26 gpios=reset:25,dc:24,led:18 rotate=90 it work but not load when start

Thank you

notro, 2015-10-14:
If you use Raspian Jessie (the latest) it uses systemd, so try this: https://github.com/notro/fbtft/wiki#make-it-permanent-arch-linux

Jakkawad, 2015-10-14:
thank you i will try ,and it work for 3.5 inch touch screen? i order i from eBay .

notro, 2015-10-14:
> and it work for 3.5 inch touch screen?

I don't know

Jakkawad, 2015-10-15:
[image]
[image]
[image]

notro, 2015-10-15:
fbtft_device apparently is loaded before spi-bcm2835 so there's no spi bus 0.
Try adding spi-bcm2835 before fbtft_device.

Jakkawad, 2015-10-15:
thank you,i will try now

Jakkawad, 2015-10-15:
it work!!!  THANK YOU

closed 2024-09-29 by notro