notro/rpi-firmware #30: Update to Raspberry Pi 3

DatN3xus, 2016-03-05:
rpi-firmware isn't compatible with Raspberry Pi 3

DatN3xus, 2016-03-05:
Raspberry Pi doesn't start up after rpi-update

AndrewFromMelbourne, 2016-03-07:
The firmware is this repository is very old. The Raspberry Pi Foundation kernel now contains the fbtft driver, so it is no longer necessary to download the kernel/firmware from here.

DatN3xus, 2016-03-12:
Oh, thank you for the update sir! :)
But how do I activate it? The display doesn't run with the latest raspbian and kernel... :(

AndrewFromMelbourne, 2016-03-12:
It depends on the display that you have attached to your Raspberry Pi. I would suggest asking for help in the Raspberry Pi forums - https://www.raspberrypi.org/forums/

Ryanphs, 2016-08-05:
Hello All. Did any of you ever get this working? If so please let me know how, I'd be really grateful.

AndrewFromMelbourne, 2016-08-05:
@Ryanphs. You are more likely to get the information you need on the Raspberry Pi forums. You will need to know which display you are using, as there are a number of different varieties. It will also help if you say what you would like to do with the display, as once the display is set up correctly it will be just black. You can use theses display for a variety of purposes, but setting them up to do what you want can be a little tricky.

closed 2019-01-04 by 3lue